aica-technology/control-libraries #17: Addition and subtraction operators in state representation

buschbapti, 2023-01-19:
While we have already taken care of multiplication operators of Cartesian states in control libraries (i.e. that it's not possible to multiply a wrench by an acceleration), it's still possible to do all possible combinations of additions and subtractions.

For example, the following operation is possible
```
CartesianPose += CartesianWrench;
```

Similar to the multiplication, we should be more precise and prohibit certain combinations. Allowed operations would be:
- `CartesianState` + any
- any derived class + the same type of derived class
- any derived class + `CartesianState`

Additionally we could also allow operations like `CartesianPose + CartesianTwist` if we integrate the twist before adding it.

eeberhard, 2023-01-19:
I could look at the documentation but I just want to check my intuition. I would expect that:

The `+` operator is for combining states in the same reference frame
The `*` operator is for expressing a child state in a parent reference frame

Correct me if I'm wrong. With that in mind, I think the compatibility rules should indeed be the same for `+` as for `*`, like you suggest 👍

domire8, 2023-01-19:
Yes indeed, you are correct. I think the way it will be implemented is the following, in addition to the rules above:
- an addition including a CartesianState will always result in a CartesianState (unless its the +=)
- Additions and subtractions with implicit derivation or integration won't be allowed

buschbapti, 2023-01-19:
Correct. Some considerations while working on it:
- As opposed to the `*` a `CartesianState` + any derived should result in a `CartesianState` no matter the order of operation. This is to avoid the loss of information
- Implicit derivation or integration is actually removed as you can't say in which order it should go, .e.g `CartesianPose + CartesianTwist` could be converted to `CartesianPose + CartesianPose` or `CartesianTwist + CartesianTwist` which is highly ambiguous.

eeberhard, 2023-01-20:
> Additions and subtractions with implicit derivation or integration won't be allowed

I always prefer explicit behavior to implicit behavior, so I agree with forbidding those combinations (though I don't think the order is ambiguous. Anyway the problem with implicit integration / derivation is that it's always with unit time, so has limited value in practice.

> you can't say in which order it should go, .e.g `CartesianPose + CartesianTwist` could be converted to `CartesianPose + CartesianPose` or `CartesianTwist + CartesianTwist` which is highly ambiguous.
I don't see this as ambiguous, but rather just as bizarre; I would say usually the left operand defines the context, much like the `+=` operator. The most "reasonable" example is `Pose + Twist`, which could feasibly be a unit integration of the twist into pose. To say that `Pose + Twist` could by `Pose / 1s + Twist` is much less reasonable. But overall the point is that these implicit operations don't make sense and so should not be allowed

eeberhard, 2023-01-24:
I think to address `CartesianState = CartesianTwist - CartesianState` we have to think of it as `CartesianState = CartesianTwist + (-CartesianState)`, which means first to consider what `-CartesianState` means. To me, it's quite clear;

```c++
CartesianState CartesianState::operator-() const {
  return this->inverse();
}
```

While I don't know if anyone would really want to do the suggested operation, I think it should be permitted and behave as described.

buschbapti, 2023-01-24:
In fact, apart from the changing of reference frames the calculation is not that different:

```cpp
CartesianState CartesianState::inverse() const {
  CartesianState result(*this);
  // inverse name and reference frame
  std::string ref = result.get_reference_frame();
  result.set_reference_frame(result.get_name());
  result.set_name(ref);
  // intermediate variables for f_S_b
  Eigen::Vector3d f_P_b = this->get_position();
  Eigen::Quaterniond f_R_b = this->get_orientation();
  Eigen::Vector3d f_v_b = this->get_linear_velocity();
  Eigen::Vector3d f_omega_b = this->get_angular_velocity();
  Eigen::Vector3d f_a_b = this->get_linear_acceleration();
  Eigen::Vector3d f_alpha_b = this->get_angular_acceleration();
  // computation for b_S_f
  Eigen::Quaterniond b_R_f = f_R_b.conjugate();
  Eigen::Vector3d b_P_f = b_R_f * (-f_P_b);
  Eigen::Vector3d b_v_f = b_R_f * (-f_v_b);
  Eigen::Vector3d b_omega_f = b_R_f * (-f_omega_b);
  Eigen::Vector3d b_a_f = b_R_f * f_a_b;        // not sure if minus is needed
  Eigen::Vector3d b_alpha_f = b_R_f * f_alpha_b;// no minus for sure
  // wrench
  //TODO
  // collect the results
  result.set_position(b_P_f);
  result.set_orientation(b_R_f);
  result.set_linear_velocity(b_v_f);
  result.set_angular_velocity(b_omega_f);
  result.set_linear_acceleration(b_a_f);
  result.set_angular_acceleration(b_alpha_f);
  return result;
}
```

There is still the rotation for each of those vectors though. But the conjugate and minus are there (apart for the acceleration).

eeberhard, 2023-01-24:
You're right, I oversimplified it and was wrong.

My line of thinking was that that the negation of a state should be equivalent to its reflection about the origin, which what I intuitively thought was the same as inversion. But the inverse is of course changing the frame of reference, and as a result the apparent state vectors are all rotated instead of just reflected.

So concerning reflections in the same reference frame, the `CartesianState CartesianState::operator-() const` would simply negate all vectors and conjugate the orientation.

In any case we should strive to be internally consistent, so that:
- `state - state == state + (-state) == identity/zero`

I wold still maintain that `CartesianState = CartesianTwist - CartesianState` and co are generally permissible. We can also think of it as `CartesianState = CartesianState(CartesianTwist) - CartesianState`. For that we just need to make sure upcasting / slicing is as safe as possible. 

On a tangential note, it also stands to reason that the following should be true:
- `state * state.inverse() == identity/zero`
(except that the latter operation also reduces the state to "reference frame in reference frame")
But, it's not quite right at the moment, and thinking about all this has inspired me to fix the inverse and transform operations once and for all. Not tonight though 😅

buschbapti, 2023-01-24:
Right. Makes sense and I definitely agree with the checks you also mention. In fact those make interesting test sets. I like the idea of the `-` operator which would also simplify the subtraction operators. Good stuff.

buschbapti, 2023-01-26:
Now that both #23 and #33 have been merged, we can close this issue.

eeberhard, 2023-01-27:
@domire8 correctly noticed that the revised and simplified `-()` operation now introduces problems with the state transformation. The reasoning is as follows:

To begin, for representations of orientation, the quaternions `q = -q`, `[w x y z] = [-w -x -y -z]` are equivalent. They are on opposite ends of the SO(3) hypersphere (they are on different hemispheres) but their projection into 3D is equivalent.

For some quaternion `q`, the operation `q.conjugate() = [w -x -y -z]` is used to invert the quaternion. When used to represent rotation, it means that the direction of rotation is reversed. This is the intended behavior of the operation.

So, indeed, `a + b` is the same as `a + (-b)` for a pose. However, it is not the same for a _state_. When dealing with derivatives like twist, the quaternion is used and manipulated in ways that do not always represent a rotation. Specifically, derivatives use the vector (imaginary) part of the quaternion to represent angular displacements.

For the representation of a single twist, the quaternion hemisphere also doesn't matter; a positive rotation about a positive axis is equivalent to a negative rotation about a negative axis. But for operations involving two states, it is imperative that the two representations are on the same hemisphere. Put simply, the choice of representation needs to be consistent in every operation.

By using the `conjugate()` method from `Eigen` in the `-()` operator, we are not just reversing the represented orientation but also switching the hemisphere. In the 4-dimensional vector space, the result of `q.conjugate() = [w -x -y -z]` is further away from the original vector `[w x y z]` than the alternative `[-w x y z]`. Reversing the scalar (real) term keeps the quaternion on the same hemisphere and is rotationally equivalent to the other conjugation. Because `q = -q` for rotations, `q.conjugate() = -q.conjugate()`.

In summary, the operator `-()` should not use the Eigen `conjugate()` method directly but instead ensure the resulting quaternion has only the real scalar part reversed:

`q.w() = -q.w()`

eeberhard, 2023-01-27:
Upon further consideration, while the above is technically correct, the quaternion conjugate being in the other hemisphere is actually the internally consistent behavior.

Let me represent the hemisphere with a superscript:

$$
-q^+ = q^-
$$

A conjugation flips the hemisphere:

$$
(q^+)^* = q_c^-
$$

And the multiplication of a quaternion in one hemisphere with the conjugate of another quaternion originally in the same hemisphere results in the original hemisphere.

$$
q_1^+ * (q_2^+)^* = q_3^+
$$

The addition / multiplication operators for e.g. `q1 + q2` currently rectify `q2` to be in the same hemisphere as `q1` _before_ the operation. But, if `q2` represents a conjugate, then it is "intentionally" in the other hemisphere, such that the product would be back in the original hemisphere. 

So, let's say we try to do `q1 + q2c`, where `q2c` is a conjugate of a consistently represented `q2` and therefore falls in the other hemisphere:

$$
(q_2^+)^* = q_{2c}^-
$$

Now, the operator `+` checks the operands and flips `q2c` for not being in the right hemisphere. The result? 

$$
q_1^+ * -(q_{2c}^-) = q_3^-
$$

The output is now in a different hemisphere. Again, this is a perfectly valid orientation and equivalent to the negative case. But as noted above, it is no longer a consistent representation with respect to the original state, and so derivations / integrations will have inconsistent behavior.

The solution is to check and enforce the hemisphere _after_ the quaternion product, not before.

domire8, 2023-01-27:
Thank you for writing that down cleanly for posterity! It makes a lot of sense to me.

domire8, 2023-01-28:
Hoping that this is now completely resolved, I will close this issue again. Next up is #18.

closed 2023-01-28 by domire8

buschbapti, 2023-01-28:
Beautiful. Thanks @eeberhard for the very detailed explanation!