aica-technology/control-libraries #20: StateType should not be part of swap but tied to the class type

domire8, 2023-01-19:
While implementing other operators for issue #17, @buschbapti detected the following bug:
The multiplication of a CartesianState with a CartesianPose will correctly give a CartesianPose object, but it will have type Cartesian state.
```
auto res = CartesianState() * CartesianPose();
// results in res.get_type() == StateType::CARTESIAN_STATE even though its a CartesianPose object
```
This is due to the default copy constructor in `CartesianState`.

I believe, like it is the case for the State, that each object should have its own corresponding `StateType` and not hold any other type there (a `CartesianState` is always `CARTESIAN_STATE` and not `CARTESIAN_POSE` or similar). We therefore have to add those copy constructors manually, as well as remove
```
this->type_ = state.type_;
```
from the `State::swap` method.

There were some questions marks if that is a good move, for example if you want to store derived Cartesian state objects in a `std::vector<CartesianState>`, you will loose the information that they were another type before the casting. However, I believe that if you want to store Cartesian states of different types in a list, map, whatever, you should use smart pointers instead! Tying the state type to the state class is in my opinion a good move for usability and transparency. The `StateType` remains useful as a programmatical way in cpp to find out what type of object you have as you can't just do `isinstance` easily.

What do you think @eeberhard?

buschbapti, 2023-01-19:
I think this goes in the good direction to have highly consistent behaviors for those state. Indeed if we allow this type of constructions we might end up in some weird situations where a pose could have a state type twist when chaining operations. It makes the type a non mutable class attribute. In fact, we could go further by declaring it as `const`.

eeberhard, 2023-01-20:
I completely agree. I remember this caused me a lot of headaches in certain unit tests a long time ago (I think when making clproto) because the state type was not what I initially expected. For concrete instances, the state type should _always_ reflect the instantiated type. A raw CartesianState should always have type cartesian state. As you say, the type is still a very useful property for introspection during pointer injection and variadic object handling.

> In fact, we could go further by declaring it as `const`.

That's a great idea to really solidify the design intention. For this we will need to assign the correct state type during constructor member initialization. But currently some cases will still require the assignment in the constructor / operator body; for example, every instance of `set_type`.

I would suggest the following actions:
- [ ] promote `type_` to a protected field
- [ ] make `type_` const
- [ ] remove `set_type` method entirely
- [ ] replace all instances of `set_type` in derived constructors with `type_()` in the member initializer list.

eeberhard, 2023-01-20:
As Dominic pointed out, It is not possible for derived constructors to initialize the parent property in the member initializer list, since the parent constructor would have already been invoked at that point.
https://stackoverflow.com/questions/2290733/initialize-parents-protected-members-with-initialization-list-c

So, the revised approach would be:

- [ ] make `type_` const (keep it private)
- [ ] add protected State constructor that takes a `type` argument.
- [ ] remove `set_type` method entirely
- [ ] replace all instances of `set_type` in derived constructors with `State(type)` in the member initializer list

domire8, 2023-01-24:
I need to reopen this issue because the copy construction does not work as expected. `CartesianState(pose)` still has type `CARTESIAN_POSE`, which is incorrect, it should be `CARTESIAN_STATE

domire8, 2023-01-27:
Closing this again now. Follow up issue is #35. Slicing is still something we need to pay attention to in the future

closed 2023-01-27 by domire8