aica-technology/control-libraries #30: Fix transform operation in CartesianState

eeberhard, 2023-01-24:
The transformation operator `*` should correctly respect acceleration and wrench. For starting reference, see this excellent PR draft: https://github.com/epfl-lasa/control-libraries/pull/164

eeberhard, 2023-02-14:
Force[^1] cannot be transformed in the same way as position, velocity or acceleration. I will try to justify this concept with intuitive reasoning.

### 1. Force is an applied property, not an inherent property

Spatial properties are inherent to a body.  It is always possible to choose a point on a body (the frame) and expressed its spatial properties (position, orientation, linear and angular velocity and acceleration) relative to an observer frame (the reference frame).  

Simply put, a frame will always "have" a relative position, velocity, etc. The value itself just depends on the observer.

In contrast, a frame cannot "have" a force. It is not possible to pick a random point and ask what force it has relative to an observer. Instead, a force is something that is _applied to_ a frame.

### 2. Force is not relative

Spatial properties are always defined relative to an observer frame (the reference frame). A frame that is stationary in one reference frame can be moving or accelerating relative to another reference frame. 

If someone throws a tennis ball out of a car window while the car is driving, then the speed of the ball relative to the ground is the speed of the throw combined with the speed of the car. 

Force is not relative. If a frame has no force applied to it, then no choice of observer frame will change that.

The tennis ball in the previous example has a downwards force acting on it because of gravity; say 1 Newton. The car meanwhile experiences 10000 Newtons of gravitational force against the road in addition to forward driving forces from the wheels. Does this mean that the tennis ball experiences 10001N of force relative to the ground? Certainly not.

### 3. Forces do not sum in series

It is clear that any forces applied to two arbitrary, unconnected, "free-floating" frames will not be combined; the tennis ball and the car have different applied forces because they aren't rigidly connected. But, what about two frames that _are_ connected in a linkage, such as the transform tree of a robot manipulator? Surely then the force of linked frames will be combined.

Imagine a simple robot with two linear actuators connected end to end. Each actuator can extend from 0m to 1m with 10N of extension force. The tip of the robot can therefore extend up to 2m from the base, because the displacement from each actuator is combined. If the robot now pushes against a force sensor with maximal extension force, what will the force sensor measure? Is it 20N because of the two 10N actuators working together? Unfortunately not; the measured force will only be 10N[^2].

[^1]: I am using "force" interchangeably with "wrench" just for improved readability and intuition.
[^2]: This is because of laws of conservation of energy. It would be very impolite to break them. If the actuators are arranged in parallel, they can arrive at 20N, but of course the stroke length is reduced from 2m to 1m, so the total work performed increases proportionally and not exponentially

eeberhard, 2023-02-17:
I wrote a long followup comment the other day but I think I never pressed submit, so I lost it at some point when the page refreshed. It had a lot of words and equations and so on, but the short version is:

Our expression of wrench is equivalent to the reading of force torque sensor at the frame, expressed in (aligned with) the reference frame. So, the state $^W\mathcal{C}_A$ has wrench $^W\mathcal{F}_A$, measured at A and expressed in coordinate frame W. For the product of two states $^W\mathcal{C}_A * ^A\mathcal{C}_B = ^W\mathcal{C}_B$, the resulting wrench $^W\mathcal{F}_B$ must have the same magnitude as $^A\mathcal{F}_B$, except that it is rotated to align with coordinate system W instead of A. The intermediate wrench at A ($^W\mathcal{F}_A$) is discarded.

The inverse also doesn't make sense if you consider that $^A\mathcal{F}_B$ is a wrench measurement at B. The inverse $^B\mathcal{F}_A$ is indeterminate without a measurement at A, unless you make a whole lot of assumptions about rigid body connections and transfer of force. If we did that, it would be internally inconsistent with the rest of the spatial inverse, which assumes that the two frames are _not_ rigidly connected. So, the inverse of the wrench should remain undefined.

domire8, 2023-02-17:
I absolutely agree with you on the final decision. I'm happy to treat the wrench differently and the way you propose is clear and comprehensible for users as well.

PR #32 (2023-01-26): Fix inverse operation in CartesianState

## Description

- #29 

After a lot of head scratching and right-hand-ruling, I was able to understand how our representation of twist and accelerations behave in the inverse case. I added a lot of specific test cases for how the various components of state interact and I think it's now in a correct state for twist and acceleration.

This issue was not the highest priority but I felt inspired to do it on my Wednesday hack day as part of the release 7.0 milestone. I was additionally motivated to better understand our transforms so that I can finally reso [... cut]
closed 2023-02-20 by eeberhard