aica-technology/control-libraries #43: Multiplication operators with a matrix in Cartesian is incoherent

buschbapti, 2023-01-28:
The multiplication operators with a `Eigen::Matrix` are not well defined in the Cartesian classes. Below is the list of concerns:
- It is not defined in the `CartesianPose` nor `CartesianState`
- It is only defined as `matrix * state` which makes sense as `state * matrix` would be a weird order. However, we still define the `state *= matrix` which technically is in the wrong order (nitpick)
- It takes as argument a `Eigen::Matrix<6, 6>` but, in fact, only the upper and lower `3x3` blocks are used in the computation

The first two comments are open for discussion, especially the first one as the definition of matrix scaling for pose is not clear. Third comment, however, should be addressed. We should not prevent someone from considering a coupling between the linear and angular parts. It should be the user responsibility to create and uncoupled gain matrix by defining `3x3` blocks.

buschbapti, 2023-01-29:
For consistency, I propose to accept the matrix multiplication with a `CartesianPose`. Obviously it will be a 7x7 matrix. Second point is very nitpick and I think it is clear enough that the `*=` should result in a correct operation.

buschbapti, 2023-01-29:
Looking at the `JointState`, I realized that we have also made there the method to multiply the states with a `Eigen::ArrayXd`. For consistency we might want to add it in Cartesian as well.

domire8, 2023-01-30:
Here my thoughts:
- I would not define it for pose and state, as it is, because I don't know what effect a scaling would have for the orientation. I think this is too complicated.
- We did all those operator PRs to be concise and consistent. Hence, I would remove the `*=` because it's technically just incorrect.
- Agreed that it should not split linear and angular part
- Regarding `Eigen::ArrayXd`, this is a feature that can wait IMO. We barely ever use the matrix scaling operator so this is a feature that can be added when required.

eeberhard, 2023-01-30:
> I would not define it for pose and state

I agree with Dominic there. While the `[7x7] * [7x1]` is technically feasible, linear operations on non-linear space SO(3) are not sensible. Just because we can doesn't mean we should, and therefore we can keep those methods intentionally undefined.

The second point is debatable and I would err on the side of caution. If our state variables were normal vectors, then front/back multiplication of matrices would be expected to give different results if the matrix is not diagonal, e.g. $A * s = (s^T * A^T)^T$. Our state variables are not normal vectors though, and so we don't really think of `state.transpose()` in that sense. So the above would be equivalent to $A * s = s * A^T$. In other words, if we support both `state * matrix` and `matrix * state`, then I would expect the input matrix to be transposed in one or the other case. This can be very confusing very quickly since we don't know which is the "normal" case. But, if we only allow one direction (`matrix * state`) then it is unambiguous. This would imply removing the `*=` matrix operator, which I think is fine because those cases can just be expanded, which is more readable in the end.

The third point is fair for cases where linear and angular terms are intentionally convolved with non-diagonal matrices, so I agree it can be updated. I also don't think this would affect any of our current use-cases so it should be safe to do.

eeberhard, 2023-01-30:
And for the last point, element-wise scaling, the user can supply a diagonal matrix if necessary. We should only add convenience methods when we see lots of repeated use or requests, otherwise our interface can get bloated.

buschbapti, 2023-02-01:
After merging #46 this issue can now be closed.

closed 2023-02-01 by buschbapti