aica-technology/control-libraries #50: Inconsistencies in Parameter class

domire8, 2023-01-31:
I have come across several points in the `Parameter` class that bother me:
1. Empty construction is not possible (a name at least is required). This is a difference to the large majority of the other state representation classes
2. There are "empty" parameters, but the behavior is really inconsistent.
   - `get_value` will not throw an exception in case the parameter is empty, but return the value anyway
   - Copy construction does not bother about emptiness, e.g. if you copy construct from an empty parameter, you will end up with a non-empty parameter
3. There is a `T& get_value()` and a `const T& get_value() const` and I don't understand the need for both
4. The `State::initialize` method is not overridden, even though it could also be useful here

Additionally, I introduced a bug in clproto in #9:
```
template<typename ParamT>
inline state_representation::proto::Parameter encoder(const state_representation::Parameter<ParamT>& parameter) {
  state_representation::proto::Parameter message;
  *message.mutable_state() = encoder(static_cast<state_representation::State>(parameter));
  if (parameter.is_empty()) {
    return message;
  }
  return encoder<ParamT>(message, parameter);
}
```
5. If you encode an empty parameter, you only modify the `mutable_state` field of the message and hence, protobuf can't figure out what type of parameter it is.

As solutions, I propose the following:
1. Make the constructor argument `name` optional: `Parameter(const std::string& name = "")`. This will allow for constructions like `Parameter<int> param;`
2. Fix copy construction and throw an exception on `get_value` if the parameter is empty
3. Hoping for an explanation from you guys :)
4. Implement the `initialize` method for parameters
5. Here, I'm not sure what a good solution could look like...

Looking forward to hear your thoughts @eeberhard @buschbapti

domire8, 2023-02-06:
It would be great if we could discuss this in the next few days, so I have a few days to implement this in the current iteration @buschbapti @eeberhard

eeberhard, 2023-02-07:
We had an in-person discussion. The notes are

- Having an empty name with default constructor is ok, to be consistent with other states
- We _could_ forbid empty name, by throwing exception if `set_empty(false)` is called while the name is empty. However, this is not a priority
- Parameter should include `initialize()` to reset to default (empty)
- Getters should indeed be forbidden when the state is empty (in general, across all states)
- Copying an empty state should ignore data fields (or rather, if empty -> set zero) (across all states)

As a followup refactor issue, I proposed to rename the following method for clarity:
`initialize()` -> `reset()`

domire8, 2023-02-07:
Thanks for updating this. The notes address all but one point - the clproto issue. The message definition of a `Parameter` is
```
message Parameter {
  State state = 1;
  ParameterValue parameter_value = 2;
}
```
and if we don't have a `parameter_value` for the parameter, we can't correctly decode it. We'll have to dig deeper into that

domire8, 2023-02-07:
Edit: The copy construction of a `Parameter` in C++ was already correct, not in Python though

eeberhard, 2023-02-08:
As a note, Parameter inherits from ParameterInterface, which has no default constructor. To me this is sensible because ParameterInterface _must_ have the parameter type on construction, and should generally be used with pointers anyway. 

The Python bindings do not bind Parameter directly but rather a ParameterContainer inheriting from ParameterInterface. For this reason (mainly because of lack of effective type assertions in Python) the construction signature for the python ParameterContainer also always requires the parameter types and cannot be empty constructed.

Just wanted to point those details out so we can agree that the ParameterInterface and the Python Parameter still require a name (and type) on construction.

domire8, 2023-02-08:
I 100% agree here, I had the same reasoning in my head but didn't take to write it down. Thanks!

domire8, 2023-02-09:
For the problem with the clproto encoders, I think I have found a solution:
```
if (parameter) {
  message.mutable_parameter_value()->mutable_int_()->set_value(parameter.get_value());
} else {
  message.mutable_parameter_value()->mutable_int_()->set_value(int());
}
```
We just set a default value for the parameter when encoding. The state itself is still empty though so the decoder will produce an empty parameter
```
if (message.state().empty()) {
  return Parameter<int>(message.state().name());
}
return Parameter<int>(message.state().name(), message.parameter_value().int_().value());
```

domire8, 2023-02-13:
Nice, after this last PR, I think I can close this issue. Thanks for your help, I'm quite happy with it!

domire8, 2023-02-13:
Oh I got too excited too early haha. Still need to throw an exception if a state is empty

domire8, 2023-02-15:
For that last point, we have a lot of situations like here
```
CartesianTwist CartesianAcceleration::operator*(const std::chrono::nanoseconds& dt) const {
  // sanity check
  if (this->is_empty()) {
    throw EmptyStateException(this->get_name() + " state is empty");
  }
  // operations
  CartesianTwist twist(this->get_name(), this->get_reference_frame());
  // convert the period to a double with the second as reference
  double period = dt.count();
  period /= 1e9;
  // convert the acceleration into a twist
  twist.set_linear_velocity(period * this->get_linear_acceleration());
  twist.set_angular_velocity(period * this->get_angular_acceleration());
  return twist;
}
```
where we do sanity checks before starting the calculations. Now that `get_linear_acceleration()` will throw the exception, this is not really needed anymore, do you agree? We waste some effort with the first few lines but I would prefer if we can just say that these sanity checks are not needed anymore as all the getters will throw. And I remember a discussion we had that if something fails anyway, it doesn't matter if we wasted some computational effort.

eeberhard, 2023-02-17:
I agree that the repeated sanity checks in many places could be reduced or simplified. To confirm what I said in a previous discussion, I think it's better for something to run more efficiently in the nominal case than in the exceptional case (e.g. by eliding assertions and checks early, and failing later instead).

So, for those cases where an appropriate exception would be thrown by the subsequent operations anyway, I'm happy if the redundant early checks are removed.

domire8, 2023-02-17:
Okay thanks for the answer, I will then close this issue and open a new one of the `throw on empty` feature. This is not inconsistent in the parameter class but something new for the whole library

closed 2023-02-17 by domire8