aica-technology/control-libraries #92: `set_zero` should result in a non-empty state

domire8, 2023-02-22:
For the Cartesian and joint states, we have the `set_zero` method. As the name suggests, it sets all fields back to zero. However, that's the only thing it does, and I believe that this operation should in a non-empty state. For me the two following things are equal:
```c++
auto one = CartesianState::Zero("test");

auto two = CartesianState("test");
two.set_zero();
```
Currently, `set_zero` is used in `reset` to set the data back to zero:
```c++
void CartesianState::set_zero() {
  data.setZero() // set all fields to zero
}

void CartesianState::reset() {
  this->State::reset();
  this->set_zero();
}
```

I think that it should be the other way round: Reset sets all fields to zero and does `State::reset` and then `set_zero` resets the state and sets it back to non-empty.
```c++
void CartesianState::reset() {
  data.setZero() // set all fields to zero
  this->State::reset();
}

void CartesianState::set_zero() {
  this->reset();
  this->set_empty(false);
}
```

eeberhard, 2023-02-23:
Actually I think I disagree here. I don't like `set_zero` calling `reset`. Even though we know what `reset` does, it's a mixed behavior to call in in `set_zero`.

A setter should update data and return a non-empty state.

The reset should reset data and return an empty state.

I would just add `set_empty(false)` to the set_zero setter, and then reset the state after setting zero in `reset`.
```cpp
void CartesianState::set_zero() {
  data.setZero();
  this->set_empty(false);
}

void CartesianState::reset() {
  this->set_zero();
  this->State::reset();
}
```

domire8, 2023-02-23:
Yes, even better :+1:

closed 2023-02-23 by domire8