aica-technology/control-libraries #96: Make State and SpatialState abstract classes

eeberhard, 2023-02-23:
As a breaking change, `State` and `SpatialState` could be defined as abstract classes, to prevent instantiation of these classes alone.

eeberhard, 2023-02-23:
A normal way of defining abstract classes is to define a pure virtual function. In the simplest case, this function can be the destructor: `virtual ~State() = 0`. However, making the bases classes fully abstract would also require the copy and swap operations of derived classes to be revised, since an abstract class cannot be copied or swapped by value. I don't really want to propose this design change so close to a release.

An alternative is to make the class "protected" instead of abstract, by marking all constructors as protected. This will also prevent explicit construction while not actually being abstract.

eeberhard, 2023-02-23:
The more I think about it, the less I like it. If we redesigned this library from scratch, it would be nice for these interface classes to be fully abstract, but making a half-way solution doesn't seem worth the effort.

@buschbapti @domire8 your opinions are welcome whether to resolve or close this issue.

domire8, 2023-02-23:
Honestly, I don't have a strong opinion. I like the idea but I think any change that we make now because we believe its a small one will further delay 7.0 and the benefit is not that big.

eeberhard, 2023-02-24:
I agree, it's not worth it for this release.

closed 2023-02-24 by eeberhard