aica-technology/control-libraries #99: Refactor Jacobian operators

domire8, 2023-02-28:
Follow operations in CartesianState and JointState to make Jacobian fully consistent

domire8, 2023-02-28:
I will make a start here and list the current functionalities of Jacobian:
1. `Jacobian Jacobian::inverse()`
2. `Jacobian Jacobian::pseudoinverse()`
3. `Matrix Jacobian::solve(Matrix)`
4. `Matrix Jacobian::solve(Matrix)`
5. `Matrix Jacobian::solve(Matrix)`

buschbapti, 2023-02-28:
Agree with your points but disagree on the name for 3. Maybe `JointVelocities Jacobian::transform(CartesianTwist)`?

domire8, 2023-02-28:
I missed the fact that `Jacobian::transpose * CartesianTwist` is not a thing so 11 can be removed without replacement

eeberhard, 2023-02-28:
Final decision
```c++
Matrix Jacobian::transpose()
Matrix Jacobian::inverse()
Matrix Jacobian::pseudoinverse()

JointTorques Jacobian::transpose(CartesianWrench)
JointVelocities Jacobian::inverse(CartesianTwist)
JointVelocities Jacobian::pseudoinverse(CartesianTwist)

Matrix Jacobian::inverse(Matrix) // instead of solve
Matrix Jacobian::pseudoinverse(Matrix)

Matrix Jacobian * Matrix
Jacobian Matrix * Jacobian // only for 6x6

CartesianTwist Jacobian * JointVelocities
```

Option we voted against:
```
JointTorques Jacobian::to_joint_space(CartesianWrench)
JointVelocities Jacobian::to_joint_space(CartesianTwist, type=PSEUDO)

CartesianTwist Jacobian::to_task_space(JointVelocities)
```

Discouraged / unsupported:
```c++
// CartesianWrench Jacobian::transpose() * CartesianWrench -> use the following instead
Matrix Jacobian::transpose() * CartesianWrench.data()

// Jacobian * Jacobian -> not possible
```

domire8, 2023-03-01:
I have one last point (hopefully). The `set_reference_frame` currently takes a Cartesian pose as an argument, to change the reference frame of the jacobian. This is hugely different from the behavior in Cartesian state and spatial state. I would almost prefer that we have `set_reference_frame(const std::string& name)` that just sets it and then we have (as is) the operator with a Cartesian pose that will transform the jacobian to a different reference frame. If that's not enough, we could have something like `change_reference_frame(CartesianPose)` but I'm not sure that this is necessary.

domire8, 2023-03-02:
Thanks for all the help and reviews

closed 2023-03-02 by domire8