aica-technology/control-libraries #146: Construct and convert CartesianPose from rotation or transformation matrices

eeberhard, 2023-10-20:
It would be convenient to be able to construct a `CartesianPose` object from a 4x4 transformation object. Similarly, it would be convenient to have a direct method `get_transformation_matrix()` or similar to return the pose as a 4x4 transformation matrix.

The 3x3 rotation matrix could be handled similarly, though for construction it would take either a 3vector for position or set position to 0.

This feature would make operations with other standard representations even easier.

domire8, 2025-01-07:
1. `get_transformation_matrix()` already exists as function of `CartesianState`
2. Adding constructors like 
   ``` 
   explicit CartesianPose(const std::string& name, const Eigen::Matrix4d& transformation_matrix, const std::string& reference = "world");
   ```
   is not as easy as it looks because it conflicts with existing constructors that Eigen is not able to resolve. We already have
   ```
   explicit CartesianPose(const std::string& name, const Eigen::Vector3d& position, const std::string& reference = "world");
   ```
   so then everything thats something like `CartesianPose("A", 10 * Eigen::Vector3d::Random())` becomes ambiguous to the compiler.
3. One way to avoid this is to make a static method `CartesianPose CartesianPose::from_transformation_matrix`, it we want that

eeberhard, 2025-01-09:
> get_transformation_matrix() already exists as function of CartesianState

Since 4 years no less, how lovely!

Option 3 with a static method could be nice 👍

closed 2025-01-10 by domire8