aica-technology/control-libraries #151: Incorrect calculation of difference of Cartesian states

domire8, 2023-11-13:
https://github.com/aica-technology/control-libraries/blob/8b3ae5bff8a586b150900894e73b4268fea83ceb/source/state_representation/src/space/cartesian/CartesianState.cpp#L752

There might be a problem with this line. See this situation:
[image]

If you do `marker_beta - marker_alpha` in Python, you get
[image]
which is correct for the position part but not for the orientation. It should be 
```
orientation: (0.707107, -0.707107, 0, 0) <=> theta: 1.5708, axis: (-1, 0, 0)
```
in my opinion

@eeberhard

domire8, 2023-11-13:
Okay no, this is actually correct I guess. To go from alpha to beta you need a rotation of 90deg around the position world X axis.

But then, what you proposed for the path attractor
```
segment = end - start
self._target = sr.CartesianState(start + self._t * segment)
```
does not seem to work because the target starts rotating downwards (y axis goes down instead of up) when going to marker beta

domire8, 2023-11-13:
The issue still targets a relevant line and I'm not even sure anymore if we should be doing this operation like that. If we compare what we do with the orientation in the `*=` and `+=` operator, it's exactly the same but what we expect from it is not exactly the same.

In the addition operator, we are already checking that the two states have the same reference frame. Shouldn't we transform the rotation that we want to `add` into the reference frame of the first state first and then add the rotation? It would be like an implicit reference frame change for the orientation part?

eeberhard, 2023-11-13:
>There might be a problem with this line

The line you mentioned negates all the coefficients, so the quaternion would represent the same orientation.

Thinking about it more, it's true that "vector" notation for this kind of operation just won't work. It's partly because of commutativity and partly because of reference frames.

If you then do `start + t * (end - start)` for t = 1, it would be `start + end - start` which feels like it should be equivalent to `end` but is not actually true because of how successive orientations commute.

The other thing is that `end - start` would be the same as `end + start.inverse()`, so it would be a rotation from base frame to `end` followed by a rotation from `start` to base frame, but you are right that we want to be rotating along the segment (i.e. from start to end), not from start to base or base to start.

So let's stick with `start.inverse() * end` as correct way to achieve this operation. The addition / subtraction operators are still useful but have to be used carefully and intentionally

closed 2023-11-13 by domire8