aica-technology/control-libraries #159: Revamp build structure

domire8, 2024-03-04:
Now that #140 is closed, I would still like to tackle the rest of the build system in this repo (low priority). To me, the next point to resolve would be the automated testing on the CI. Currently, we are using the "old" `development-dependencies` to build and run tests on PR, see [here](https://github.com/aica-technology/control-libraries/blob/main/.github/actions/build-test/Dockerfile). It would be great if we can move away from this and use the same Dockerfile that the CI now uses to build the image.

The main disadvantage of this Dockerfile is that it takes long to build with the dependencies so we need a caching functionality.

@LouisBrunner what are your thoughts on this? any suggestions on what actions we could use to have the "lightest" CI possible.

LouisBrunner, 2024-03-04:
We could look at using Github cache in the CI? We have heavy caching in the Dockerfile already so it would be mostly CI work.

Example: https://depot.dev/blog/docker-layer-caching-in-github-actions#docker-layer-caching-in-github-actions

domire8, 2024-03-04:
Yeah, paying depot for an open source repo that is not heavily under development anymore seems a bit overkill, the monthly subscriptions are quite expensive.

LouisBrunner, 2024-03-12:
I think we should give it a go as it should be quite easy to setup in our `.github` actions and worst-case scenario we do an intermediate image.

domire8, 2024-03-12:
Alright let's do it like that :+1:

domire8, 2024-03-25:
After discussion with @LouisBrunner and @eeberhard, these are the main take away points and planned actions:
1. The build system should be streamlined with one single Dockerfile
2. As much as possible, the CI should use caching to reduce build time
3. Proto and development dependencies images are to be removed
4. Any other leftover shell scripts and configuration files can be removed as well
5. The output of control libraries should be (per arch) a scratch image with all sources and potentially an ubuntu distro with CL installed
6. We can remove the develop as staging branch and directly merge to main. Releases will be automatically created if the CI detects package version changes.