aica-technology/control-libraries #165: Handle hard-coded locations in test urdfs

yrh012, 2024-03-13:
The UR5e URDFs, utilized in both the Python and C++ test codes, currently have hardcoded path locations. These path locations should be dynamically updated to reflect the actual file locations of the collision meshes without prior assumption on the global path.

domire8, 2024-04-08:
Haha wanted to tackle this but realized it's already done right @yrh012 ?

yrh012, 2024-04-08:
> Haha wanted to tackle this but realized it's already done right @yrh012 ?

Yes indeed! Now with the package path resolver callback, the consumer of `robot_model::Model` will have to define how the package resolves. In the case of the tests, a dummy callback can define the actual paths of the packages and that resolves the issue.

closed 2024-04-08 by domire8