aica-technology/control-libraries #211: Breaking changes in control libraries

domire8, 2025-01-08:
Knowing that some of the initiatives require breaking changes across the stack, it's possible to think about which breaking improvements we want to make in control libraries too. First and foremost, we need to look into the robot model library and think about how we can improve its usability and feature set to leverage pinocchio better and to potentially account for types of robots that are not just a chain of revolute joints.

- https://github.com/aica-technology/control-libraries/issues/205
- https://github.com/aica-technology/control-libraries/issues/200
- https://github.com/aica-technology/control-libraries/issues/126

Apart from that, we can also try to address already existing issues:
- https://github.com/aica-technology/control-libraries/issues/203
- https://github.com/aica-technology/control-libraries/issues/193
- https://github.com/aica-technology/control-libraries/issues/146

eeberhard, 2025-01-09:
I think the way #193 is implemented, it shouldn't be ABI breaking. It adds contents to the header file but doesn't change the order or visibility of existing symbols, nor change any inline or template definitions.

I agree that in general, improving the robot model class would be the most valuable breaking change.

domire8, 2025-02-05:
https://github.com/aica-technology/control-libraries/issues/222