aica-technology/control-libraries #225: State copy constructor may cause incompatible state types for derived classes

bpapaspyros, 2025-02-07:
I was going through the Trajectory classes for other developments and I realized we have this:

```cpp
State::State(const State& state) :
    std::enable_shared_from_this<State>(state),
    type_(StateType::STATE),
    name_(state.name_),
    empty_(state.empty_),
    timestamp_(state.timestamp_) {}
```

That means that `make_shared_state` (i.e., `std::make_shared<State>`) for a derived class would change the `type_` back to `StateType::STATE` instead of whatever the derived class carries. We typically don't see the effect of this because we always define copy constructors and `=` overloads in derived classes, but that may well create issues in cases where we don't have copy constructors.

I would suggest that we add a simple change of:

```cpp
State::State(const State& state) :
    std::enable_shared_from_this<State>(state),
    type_(state.get_type()),
    name_(state.name_),
    empty_(state.empty_),
    timestamp_(state.timestamp_) {}
```

Was there another reason I am missing where this would be of significance and hence has to stay in its current form?

@fyi @eeberhard @domire8

domire8, 2025-02-07:
> hence has to stay in its current form?

That is hard to say with certainty but I think if all tests pass here and in modulo with that modification, we're good.

domire8, 2025-02-10:
Hm I actually think that the point of this was that we wanted a 1:1 relation between the enum and the actual object type. Currently
```
CartesianState cart_state;
State state(cart_state);
ASSERT_EQ(state.get_type(), StateType::STATE);
```
passes. If we change this, then suddenly this `State` object would have type `StateType::CARTESIAN_STATE` which we wanted to avoid.

bpapaspyros, 2025-02-10:
> > Hm I actually think that the point of this was that we wanted a 1:1 relation between the enum and the actual object type. 
> At the same time, I would argue that you still have a `CARTESIAN_STATE` and you sliced off the derived members (indicating you could instead see the object as a CartesianState and cast it to that).

and then when you have `std::shared_ptr<State>` because you want for example to store multi-type state in a vector, you would still know what each variable really is.

domire8, 2025-02-10:
> At the same time, I would argue that you still have a CARTESIAN_STATE and you sliced off the derived members (indicating you could instead see the object as a CartesianState and cast it to that).

There is no use for an object like that though, because all other information is sliced = lost.

> store multi-type state in a vector
But that's the beauty of pointer injection, we can already store a `shared_ptr<CartesianState>` in a vector of `shared_ptr<State>`.

bpapaspyros, 2025-02-10:
yup true on both. In general I don't have strong. Then let's keep it as is

eeberhard, 2025-02-10:
> Hm I actually think that the point of this was that we wanted a 1:1 relation between the enum and the actual object type

Yes that's it, we need to make sure that the state attribute is always representative of the actual instantiated object type. 

I was initially struggling to place my thoughts on the inconsistency, but it was really the premise that was misleading me:
> That means that make_shared_state (i.e., std::make_shared<State>) for a derived class would change the type_ back to StateType::STATE instead of whatever the derived class carries. 

In the end that is actually the expected behavior:
```cpp
std::shared_ptr<State> state = std::make_shared<State>(...);
state->get_type();  // STATE
```

The intended pattern is to use dependency injection, which relies on the fact that shared_ptr<State>::get_type returns the actual object type so that we can work with it afterwards.

```cpp
std::shared_ptr<State> state = std::make_shared<CartesianState>(...);
state->get_type();  // CARTESIAN_STATE
```

Thanks for bringing it up and checking though, since such small things can easily be unintentional in other cases.

closed 2025-02-10 by eeberhard